chrisdsa/BMS #8: Réunion de lancement ( Kick off meeting)

chrisdsa, 2017-04-26:
**Réunion de lancement**  
Quand : Lundi le 1er Mai  
Ou : Telegram  
  
**Objectif** : Discuter du cahier des charges et des différents points de vue du projet  
  
**Points à discuter** : ** A remplir dans les commentaires, je vais updater la description **
Voir issue `Revoir méthodologie - Cahier des charges #3` pour la description du cahier des charges

oclarocque, 2017-04-30:
*Définir un road map global, incluant tous side projects nécessaires (e.g. simulateur de cellule)
*Discuter de la répartition des tâches de développement pour que les membres se préparent en conséquence (études des méthodologies, des logiciels, des technologies..)

bourassajulien, 2017-05-01:
* Horaire des membres et disponibilités
* Meeting entre les membres
* Meeting avec le professeur
* Gestion du budget (sur quoi on met les ressources)
* Gestion de projet (gantt? division en semaine)
* Recherche documentation et partage des connaissances
* Schéma bloc du projet

closed 2017-05-03 by chrisdsa