chrisdsa/BMS #13: Ajouter section des risques et méthode mitigation.

bourassajulien, 2017-05-03:
On peut rajouter les risques comme dans le chapitre 2.4 du livre A Systems Approach. 
On élimine ceux qui s'appliquent pas au projet et comment on va attaquer les autres.
Je ne sais pas si c'est mieux de le mettre dans le cahier des charges ou de conception par contre.

chrisdsa, 2017-05-03:
Faudrait assigner des priorités aux risques et les inclure dans les objectifs

chrisdsa, 2017-05-03:
Une autre section intéressante, chapitre 17 : **Fault Detection**  
1. Overcharge/Overvoltage
2. Over-Temperature  
3. Overcurrent  
4. Battery imbalance / Excessive Self-Discharge  
5. Internal Short Circuit Detection  
6. Detection of lithium plating  
7. Venting Detection  
8. Excessive Capacity Loss

Il parlait aussi dans le livre comment vérifier les différents périphériques. On testerais le relais, la shunt lors de la pré charge, il a des méthodes pour contre valider que les lectures sont bonne en regardant l'ensemble des autres lecture etc. Je peux ajouter ça dans les objectifs :   
- Diagnostique des périphériques / sensors  
- Contre validation des lectures (améliore la robustesse)  
Personnellement je mettrais ça dans les extras

closed 2017-05-20 by chrisdsa